nickbabcock/OhmGraphite #395: Connect to existing LibreHardwareMonitor instance?

shalak, 2023-08-16:
I'm running LHW minimized to tray 24/7. I'm using it via WMI Query to read some data used in other services.

If I understand correctly, installing OhwmGraphite will install it's own LHW instance, right? Thus I'll have two LHW running on my machine.

Is it possible to connect OhmGraphite to the LWH instance that I'm already running on my host?

nickbabcock, 2023-08-17:
That's an interesting idea. What is the concern with running two instances of LHW? 

It may be possible for the datasource that OhmGraphite exports to to be the source of truth instead of running two instances.

shalak, 2023-08-17:
> What is the concern with running two instances of LHW?

performance & unnecessary bloat

> It may be possible for the datasource that OhmGraphite exports to to be the source of truth instead of running two instances.

Yep, that's probably what I'll end up with, I just need to refactor some of my workflows. I was hoping that there might be a way to avoid it :)

nickbabcock, 2023-11-13:
I'm going to close the issue as this feature isn't planned at this time. But I'm happy to continue the discussion if there's compelling reasons.

closed 2023-11-13 by nickbabcock