nickbabcock/OhmGraphite #412: Fatal error when virtual RamDisk is running

EugeneSoftwareDeveloper, 2023-11-12:
When run [ramDisk](http://radeonramdisk.com/software_downloads.php)

[image]

[image]

When ramdisk is stoped there is no error

Fatal error. System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at System.SpanHelpers.IndexOf(Byte ByRef, Byte, Int32)
   at System.String.Ctor(SByte*)
   at System.Runtime.InteropServices.Marshal.PtrToStringAnsi(IntPtr)
   at LibreHardwareMonitor.Hardware.Storage.WindowsStorage+StorageInfo.GetString(IntPtr, UInt32)
   at LibreHardwareMonitor.Hardware.Storage.WindowsStorage+StorageInfo..ctor(Int32, IntPtr)
   at LibreHardwareMonitor.Hardware.Storage.WindowsStorage.GetStorageInfo(System.String, UInt32)
   at LibreHardwareMonitor.Hardware.Storage.AbstractStorage.CreateInstance(System.String, UInt32, UInt64, Int32, LibreHardwareMonitor.Hardware.ISettings)
   at LibreHardwareMonitor.Hardware.Storage.StorageGroup.AddHardware(LibreHardwareMonitor.Hardware.ISettings, System.Collections.Generic.Dictionary`2<UInt32,System.Collections.Generic.List`1<System.ValueTuple`2<UInt32,UInt64>>>)
   at LibreHardwareMonitor.Hardware.Storage.StorageGroup..ctor(LibreHardwareMonitor.Hardware.ISettings)
   at LibreHardwareMonitor.Hardware.Computer.AddGroups()
   at LibreHardwareMonitor.Hardware.Computer.Open()
   at OhmGraphite.SensorCollector.Start()
   at OhmGraphite.PrometheusServer.<Start>b__4_0()
   at OhmGraphite.LoggerUtils.LogAction(NLog.Logger, System.String, System.Action)
   at OhmGraphite.PrometheusServer.Start()
   at OhmGraphite.Worker+<>c__DisplayClass3_0.<ExecuteAsync>b__0()
   at System.Threading.Tasks.Task.InnerInvoke()
   at System.Threading.Tasks.Task+<>c.<.cctor>b__272_0(System.Object)
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(System.Threading.Thread, System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(System.Threading.Tasks.Task ByRef, System.Threading.Thread)
   at System.Threading.Tasks.Task.ExecuteEntryUnsafe(System.Threading.Thread)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool+WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()

nickbabcock, 2023-12-02:
Is the ramdisk included in the gui or has it been excluded?

EugeneSoftwareDeveloper, 2023-12-02:
The patition is not displayed. Only displayed hard disks and ssd.

RamDisk is not displayed (excluded). 

[image]

nickbabcock, 2023-12-16:
Do you get the same error when you run OhmGraphite interactively (in an administrator terminal)?

```pwsh
./OhmGraphite stop
./OhmGraphite run
```

EugeneSoftwareDeveloper, 2023-12-24:
> Do you get the same error when you run OhmGraphite interactively (in an administrator terminal)?
> 
> ```powershell
> ./OhmGraphite stop
> ./OhmGraphite run
> ```

Also stops after starting.
Run as admin

[image]

nickbabcock, 2023-12-30:
The screenshot is a little confusing. To run interactively, execute `OhmGraphite.exe run` (the screenshot shows `OhmGraphite.exe start`)

EugeneSoftwareDeveloper, 2024-02-01:
> The screenshot is a little confusing. To run interactively, execute `OhmGraphite.exe run` (the screenshot shows `OhmGraphite.exe start`)

c:\Program Files\OhmGraphite>OhmGraphite.exe run
2024-02-01 17:29:11.0961|INFO|OhmGraphite.Worker|Prometheus port: 2003
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/control/0 "Fan #1"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/control/1 "Fan #2"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/control/2 "Fan #3"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/control/3 "Fan #4"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/control/4 "Fan #5"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/control/5 "Fan #6"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/control/6 "Fan #7"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/temperature/0 "Chipset"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/temperature/1 "CPU"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/temperature/2 "Motherboard"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/temperature/3 "T Sensor"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/temperature/4 "VRM"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/voltage/0 "CPU Core"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/fan/0 "Chipset Fan"
2024-02-01 17:29:11.2325|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ec/current/0 "CPU"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/2 "CPU Core #1"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/3 "CPU Core #2"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/4 "CPU Core #3"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/5 "CPU Core #4"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/6 "CPU Core #5"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/7 "CPU Core #6"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/8 "CPU Core #7"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/9 "CPU Core #8"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/10 "CPU Core #9"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/11 "CPU Core #10"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/12 "CPU Core #11"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/13 "CPU Core #12"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/14 "CPU Core #13"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/15 "CPU Core #14"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/16 "CPU Core #15"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/17 "CPU Core #16"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/18 "CPU Core #17"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/19 "CPU Core #18"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/20 "CPU Core #19"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/21 "CPU Core #20"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/22 "CPU Core #21"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/23 "CPU Core #22"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/24 "CPU Core #23"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/25 "CPU Core #24"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/0 "CPU Total"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/1 "CPU Core Max"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/0 "Package"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/20 "Core #1"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/factor/0 "Core #1"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/1 "Core #1 (SMU)"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/voltage/2 "Core #1 VID"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/21 "Core #2"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/factor/1 "Core #2"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/2 "Core #2 (SMU)"
2024-02-01 17:29:11.3579|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/voltage/3 "Core #2 VID"
[... cut ...]
   at System.Threading.ExecutionContext.RunFromThreadPoolDispatchLoop(System.Threading.Thread, System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(System.Threading.Tasks.Task ByRef, System.Threading.Thread)
   at System.Threading.Tasks.Task.ExecuteEntryUnsafe(System.Threading.Thread)
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool+WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()

nickbabcock, 2024-02-02:
Thank you for posting, and your patience.

Tracing the logic, both LibreHardwareMonitor and OhmGraphite execute the same code, and while I don't know for 100%, it seems like the kernel is exposing different values to both. LibreHardwareMonitor receives either an invalid result or a false return from `DeviceIoControl` which is why it excludes it. For some reason, OhmGraphite doesn't and thinks the drive is valid (and thus fails).

I'm not sure what the next step is. Very confusing.

nickbabcock, 2024-09-14:
This should be fixed in v0.34, let me know if otherwise.

closed 2024-09-14 by nickbabcock