nickbabcock/OhmGraphite #413: Vanguard blocks inpout64x.syx

ottobaer, 2023-11-13:
It seems like Vanguard (that [Riot](https://support-valorant.riotgames.com/hc/en-us/articles/360046160933-What-is-Vanguard)  spyware thingy)  is blocking one of the kernel drivers used by OHM. 

[image]

It seems like a know issue in OHM and has something to do with Fancontrol/Gigabyte MB it seems ? 

https://github.com/Rem0o/FanControl.Releases/issues/1694 

Also it seems this is some generic kernel mode driver that is used to get direct memory access 

https://starkeblog.com/windows/kernel/driver/2021/05/15/inpoutx64.sys-windows-driver-analysis.html

Is there a way to disable the loading of this file in the first place - since it's either used for fancontrol or some Gigabyte Mainboards. 

Thank You,

nickbabcock, 2023-11-13:
Have you tried disabling the hardware with the ohmgraphite config?

```xml
<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="/motherboard/enabled" value="false" />
    <add key="/controller/enabled" value="false" />
  </appSettings>
</configuration>
```

ottobaer, 2023-11-13:
Thanks, I tried that and the error went away. 
I could even re-enable both again afterwards - I still get that message once,  but it let's me play. 

No idea what is happening here, but it seems to be fixed somehow.

closed 2023-11-13 by ottobaer