nickbabcock/OhmGraphite #427: "ohm_gpuati_factor" doesn't report data after AMD driver 23.12.1

Kybeer, 2023-12-09:
I just updated my AMD driver from 23.11.1 to 23.12.1, and notices the (Promethus) metric is no longer repoting anything "ohm_gpuati_factor"

It doesn't apperar at all if I check my endpoint http://localhost:4445/metrics. I don't know if anything can be done on OhmGraphite end, or if it's a driver thing. 

All other metrics looks fine.

nickbabcock, 2023-12-10:
- Restarting the computer may be worth trying if you haven't already.
- I'd be curious if you see the sensor listed in the GUI https://github.com/LibreHardwareMonitor/LibreHardwareMonitor#where-can-i-download-it  (if not, we're looking at an upstream issue).

Kybeer, 2023-12-10:
> * Restarting the computer may be worth trying if you haven't already.
> 
>     * I'd be curious if you see the sensor listed in the GUI https://github.com/LibreHardwareMonitor/LibreHardwareMonitor#where-can-i-download-it  (if not, we're looking at an upstream issue).

Restarting makes no difference. 

The value is visible in LibreHardwareMontor still:

[image]

nickbabcock, 2023-12-16:
Thank you for checking. The plot thickens. Can you try running OhmGraphite interactively?

```pwsh
./OhmGraphite stop
./OhmGraphite run
```

A few have reported that doing this can yield more metrics and I'm curious if this is one of those cases.

Kybeer, 2023-12-16:
> Thank you for checking. The plot thickens. Can you try running OhmGraphite interactively (in an adminstrator terminal)?
> 
> ```powershell
> ./OhmGraphite stop
> ./OhmGraphite run
> ```
> 
> A few have reported that doing this can yield more metrics and I'm curious if this is one of those cases.

Thanks for your time on this. I did what you suggested but it makes no difference. The terminal shows a log of metrics and the fps is still always 0 (line 69), just like the Prometheus endpoint. I attached the terminal output as text here. 

[ohm.txt](https://github.com/nickbabcock/OhmGraphite/files/13692347/ohm.txt)

Since this happened after a driver update perhaps its reporting FPS in a new way? On the other hand running LibreHardwareMontor still shows it correct...

nickbabcock, 2023-12-18:
To clarify, is the fps being reported as zero or not reported at all?

The sensor is being detected so that is good. My hunch is that if you were to flip on debug logging https://github.com/nickbabcock/OhmGraphite#debugging-tips there may be more log lines related to the faulty sensor.

Kybeer, 2023-12-19:
> To clarify, is the fps being reported as zero or not reported at all?
> 
> The sensor is being detected so that is good. My hunch is that if you were to flip on debug logging https://github.com/nickbabcock/OhmGraphite#debugging-tips there may be more log lines related to the faulty sensor.

Actually, `ohm_gpuati_factor{sensor="Fullscreen FPS"}` doesnt' return any data at all. The metric is not even appearing in the Prometheus endpoint (http://localhost:4445/metrics)

nickbabcock, 2023-12-30:
In the log posted earlier, I saw the following snippet:

```
Sensor added: /gpu-amd/0/factor/0 "Fullscreen FPS"
```

When you switched to debug logging, were there more lines related to `/gpu-amd/0/factor/0`?

Kybeer, 2023-12-31:
> Fullscreen FPS

I enabled Debug but I don't think there's much difference. Attaching the log file here.

nickbabcock, 2023-12-31:
The mystery deepens! I see `/gpu-amd/0/factor/0` listed in the first log but not in the second. What do you think changed?

Kybeer, 2023-12-31:
> The mystery deepens! I see `/gpu-amd/0/factor/0` listed in the first log but not in the second. What do you think changed?

I haven't changed anything myself since the driver update three weeks ago. There's been Windows updates in that time though..

nickbabcock, 2024-01-02:
Still scratching my head a bit. Thank you for your patience

Try the [OhmGraphite artifact](https://github.com/nickbabcock/OhmGraphite/actions/runs/7386059504) from the [PR that bumps to LibreHardwareMonitor 0.9.3](https://github.com/nickbabcock/OhmGraphite/pull/433)

Probably also worth double checking the [official release of LibreHardwareMonitor 0.9.3](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases/tag/v0.9.3) still shows everything correctly too.

Kybeer, 2024-01-23:
> Still scratching my head a bit. Thank you for your patience

Hi again. Driver 24.1.1 dropped today and FPS is now working again :)

[image]

nickbabcock, 2024-01-28:
That's great news, I'll go ahead and close this issue out.

closed 2024-01-28 by nickbabcock