nickbabcock/OhmGraphite #438: Publish manifest for winget

sgade, 2024-02-01:
I'd like to use [winget](https://github.com/microsoft/winget-cli) to install and update OhmGraphite.

To make OhmGraphite available, a manifest needs to be published in the corresponding [repository](https://github.com/microsoft/winget-pkgs). It's a YAML file format that can easily be bootstrapped using this tool: https://github.com/microsoft/winget-create.
I just did this locally and was able to install OhmGraphite!

A few points why I raise this issue here, though, instead of just creating a PR directly:
1. I don't what to do this without your consent, since it is your application.
2. I am not 100% sure of how the internals of both tools work together exactly. I suppose it needs to _not_ override the configuration file on upgrades, for example. I would also like to see an option to install a Windows service, but afaik that is not a feature of winget currently.
3. Since the manifests are separate by version, I think it would be easiest to automate the generation and opening of PRs using GitHub Actions.

Let me know what you think and if I can be of any help.

nickbabcock, 2024-02-02:
1. Seems like a great idea, how can I help?
2. The configuration file (`OhmGraphite.exe.config`) is meant for initial installs. There aren't breaking changes introduced to the configuration, so those updating can safely ignore it
3. I'm not quite sure what this entails. Are you saying that a release of OhmGraphite should trigger a pull request against the winget-pkgs repo? If so, sounds good.

sgade, 2024-02-06:
1. I think the best way to go about it would be that you use the `winget-create` utility to create the manifest and ensure all properties are set correctly. I used the "portable" preset but I am not sure if that is 100% correct.
2. I haven't tested it with my local install but I can do so later and report back on how winget handles unzipping with the config already existing.
3. Yes, afaik [PRs](https://github.com/microsoft/winget-pkgs/pulls) are used to create new versions. I'm new to the ecosystem myself, so I'm not sure if automating that is feasible and allowed but it would be nice to upgrade to newer versions via the package manager instead of doing it manually (which honestly I never did on my old machine).

There are a few packages which seem to only have manifests for major versions and there are some with a manifest for each minor. I guess its up to you if you want to also create a PR on every patch version as well, and how you would handle that with your versioning scheme.

sgade, 2024-02-06:
What my experimentation yielded are the manifest files (in the structure `n\NickBabcock\OhmGraphite\0.32.0\`) that I'll attach to this post.
[NickBabcock.OhmGraphite.installer.yaml](https://github.com/nickbabcock/OhmGraphite/files/14184044/NickBabcock.OhmGraphite.installer.yaml.txt)
[NickBabcock.OhmGraphite.locale.en-US.yaml](https://github.com/nickbabcock/OhmGraphite/files/14184045/NickBabcock.OhmGraphite.locale.en-US.yaml.txt)
[NickBabcock.OhmGraphite.yaml](https://github.com/nickbabcock/OhmGraphite/files/14184046/NickBabcock.OhmGraphite.yaml.txt)

The program is installed into `~\AppData\Local\Microsoft\WinGet\Packages\NickBabcock.OhmGraphite__DefaultSource`. I think that the `__DefaultSource` suffix will be dropped if installed from the official packages mirror.

I tested out creating a fake 0.33.0 version and running `winget upgrade`. This gives the following warning, I suppose because the configuration file has been changed:
```
Extracting archive...
Successfully extracted archive
Starting package install...
Unable to remove Portable package as it has been modified; to override this check use --force
```

To be honest, I find it a bit hard to see which options we could pass to ignore that file on upgrades. Once we figure out the correct manifest defintions, a GitHub bot using GitHub Actions could trigger a new PR whenever you push a tag/create a release with all the required information.

nickbabcock, 2024-02-15:
Thank you for looking into it. I was hoping this could be an easy win, but it looks like this is more complicated than anticipated.

Seems like I'll need to set aside some time to dive into the winget ecosystem.

sgade, 2024-02-15:
It is a bit opaque, to be honest.

Do you have any recommendations on how to keep the app updates otherwise? Package managers are very nice because they make it very painless to do.

nickbabcock, 2025-10-23:
Yeah publishing to winget should really be integrated into OhmGraphite directly so that when I make a release, the package gets automatically published! I could have swore I saw a tool recently that was pretty slick to automate this, but I seem to have forgotten it.

Otherwise I'll want to look into something like `vedantmgoyal9/winget-releaser`