nickbabcock/OhmGraphite #443: Failed to determine location for extracting embedded files. DOTNET_BUNDLE_EXTRACT_BASE_DIR is not set, and a read-write cache directory couldn't be created.

ramonsmits, 2024-03-15:
The service crashes at startup with the following message. However, after manually starting the service from the service control manager the service starts without any issues:

```txt
Description: A .NET application failed.
Application: OhmGraphite.exe
Path: C:\tools\OhmGraphite-0.31.0\OhmGraphite.exe
Message: Failure processing application bundle.
Failed to determine location for extracting embedded files.
DOTNET_BUNDLE_EXTRACT_BASE_DIR is not set, and a read-write cache directory couldn't be created.
```

nickbabcock, 2024-03-16:
Interesting, looking at some docs: https://learn.microsoft.com/en-us/dotnet/core/deploying/single-file/overview?tabs=cli

> If running on Windows, the files are extracted to a directory under `%TEMP%/.net`

So perhaps `%TEMP%/net` did not exist and was not writeable by the given user.

ramonsmits, 2024-03-18:
> So perhaps `%TEMP%/net` did not exist and was not writeable by the given user.

That is interesting. I have a RAM disk for `%TEMP%`. Modified system envvar's to point to a physical disk and that resolved the issue.

nickbabcock, 2024-03-22:
Glad to hear it's fixed now.

closed 2024-03-22 by nickbabcock