nickbabcock/OhmGraphite #444: incorrect battery values

FelixChial, 2024-03-22:
LHM 0.9.3 itself reports correct values
but Graphite 0.32.0 just stays at 99%, 13v, 37.4Wh, etc

[image]
[image]

dunno if its relevant but it runs as a service with prometheus output
values change sometimes but they stay the same for days

prometheus config
```yml
global:
  scrape_interval: 15s
  scrape_timeout: 5s

scrape_configs:
  - job_name: 'ohm-cobra'
    static_configs:
      - targets: ['192.168.191.103:4445']
```

graphite config
```xml
﻿<?xml version="1.0" encoding="utf-8" ?>
<configuration>
  <appSettings>
    <add key="type" value="prometheus" />
    <add key="prometheus_port" value="4445" />

    <!-- This is the host that OhmGraphite listens on.
         `*` means that it will listen on all interfaces.
         Consider restricting to a given IP address -->
    <add key="prometheus_host" value="192.168.191.103" />
    <add key="prometheus_path" value="metrics/" />
  </appSettings>
</configuration>
```

FelixChial, 2024-03-22:
it seems that values get stuck if server starts while connected
ive restarted the service while disconnected at 88% charge and it seems to work now
[image]

nickbabcock, 2024-03-25:
Do you want to try the latest [nightly version](https://github.com/nickbabcock/OhmGraphite/actions/runs/8413727133/artifacts/1354224808) of OhmGraphite? It adds trace logging so that you can see the exact values reported from LibreHardwareMonitor (see this comment: https://github.com/nickbabcock/OhmGraphite/issues/440#issuecomment-1986864013)

I wonder if it is the same problem effecting #440. If you change the `scrape_interval` to `1s`, is the problem resolved?

FelixChial, 2024-03-25:
changing `scrape_interval` to `1s` didnt help
will try nighty build, sure

FelixChial, 2024-03-26:
so for some reason simply using nightly resolves the issue, even without tracing
for sanity ive checked release again and the issue is still present 🤔 
interesting that release reports 13v Battery Voltage while nightly reports 12.9v

nickbabcock, 2024-03-27:
That's odd, I'm not sure what recent change would be influencing that. I can look to release an official new version soon.

closed 2024-04-10 by FelixChial