nickbabcock/OhmGraphite #451: Integrated GPU sensors displaces Discrete GPU sensors

meandrii, 2024-09-09:
Having two GPUs Integrate + Discrete, cause software to pull tempurate from Integrated on the Discrete Graph

I disabled Integrated AMD GPU in Device Manager and looks like graphite started to pull correct data; take a look at the graph
Orange arrow indicates the moment of disabling integrated graphics

I use latest version of software with default config, except Prometheus integration.
My hardware is:
CPU:7800X3D
GPU:RX6950XT

[image]
[image]

nickbabcock, 2024-09-09:
Thanks for the bug report. Can you try out the latest nightly build of LibreHardwareMonitor (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor?tab=readme-ov-file#where-can-i-download-it)

That'll let us know where the bug is coming from.

meandrii, 2024-09-09:
New build is on the right, and looks good to me.
[image]

nickbabcock, 2024-09-10:
Nice, can you try out the latest build of ohmgraphite? (it's a zip file found here: https://github.com/nickbabcock/OhmGraphite/actions/runs/10783732414?pr=452 )

If it works, I'll cut a new version of OhmGraphite in the next couple days.

meandrii, 2024-09-10:
I think there are a problem with powe nowr, freq and temperature for integrated not correct
And i am not sure for Discrete
[image]
[image]
[image]
[image]

nickbabcock, 2024-09-10:
Ah this might be a dashboard problem. I think I know what the problem is, but if you want to run a quick test:

Update the wattage dashboard so the metric is includes the hardware instance:

```
ohm_gpuati_watts{instance="$instance",hardware="$gpu"}
```

Does that solve the problem? You'd have to do it for each graph.

This should make it so the GPU metrics aren't stepping on each others toes.

If this works, I'll update the dashboard templates.

PR #453 (2024-09-14): Better multi-GPU and AMD prometheus dashboards
GPUs need to be differentiated just like HDD. A few nvidia gpu graphs lacked an amd equivalent.

Closes #451
closed 2024-09-14 by nickbabcock