nickbabcock/OhmGraphite #458: Install stops on 'Network Utilization'

ernestodiaztech, 2024-10-12:
I ran into this issue maybe a year ago and I was able to solve it, but I didn't write down how I solved it, and now I'm stuck trying to fix it again but haven't had any luck. I remember it being something simple.

When installing ohmgraphite, it gets stuck at Network Utilization. I've tried it on 4 different computers with the same results.

`2024-10-12 10:13:17.1086|INFO|OhmGraphite.Worker|Prometheus port: 4445
2024-10-12 10:13:18.8108|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/2 "CPU Core #1 Thread #1"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/3 "CPU Core #1 Thread #2"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/4 "CPU Core #2 Thread #1"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/5 "CPU Core #2 Thread #2"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/6 "CPU Core #3 Thread #1"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/7 "CPU Core #3 Thread #2"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/8 "CPU Core #4 Thread #1"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/9 "CPU Core #4 Thread #2"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/0 "CPU Total"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/1 "CPU Core Max"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/0 "CPU Core #1"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/1 "CPU Core #2"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/2 "CPU Core #3"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/3 "CPU Core #4"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/4 "CPU Package"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/5 "CPU Core #1 Distance to TjMax"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/6 "CPU Core #2 Distance to TjMax"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/7 "CPU Core #3 Distance to TjMax"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/8 "CPU Core #4 Distance to TjMax"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/9 "Core Max"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/temperature/10 "Core Average"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/1 "CPU Core #1"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/2 "CPU Core #2"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/3 "CPU Core #3"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/4 "CPU Core #4"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/0 "CPU Package"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/1 "CPU Cores"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/3 "CPU Memory"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/power/4 "CPU Platform"
2024-10-12 10:13:18.8300|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/0 "CPU Core"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/1 "CPU Core #1"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/2 "CPU Core #2"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/3 "CPU Core #3"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/voltage/4 "CPU Core #4"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/clock/0 "Bus Speed"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/0 "Memory Used"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/1 "Memory Available"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/0 "Memory"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/2 "Virtual Memory Used"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/3 "Virtual Memory Available"
2024-10-12 10:13:18.8453|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/1 "Virtual Memory"
2024-10-12 10:13:18.8777|INFO|OhmGraphite.SensorCollector|Sensor added: /gpu-intel-integrated/\\?\PCI#VEN_8086&DEV_1912&SUBSYS_06B91028&REV_06#3&11583659&0&10#{1ca05180-a699-450a-9a0c-de4fbe3ddd89}/power/0 "GPU Power"
2024-10-12 10:13:19.7999|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/0 "Used Space"
2024-10-12 10:13:19.7999|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/31 "Read Activity"
2024-10-12 10:13:19.7999|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/32 "Write Activity"
2024-10-12 10:13:19.7999|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/33 "Total Activity"
2024-10-12 10:13:19.7999|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/throughput/34 "Read Rate"
2024-10-12 10:13:19.7999|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/throughput/35 "Write Rate"
2024-10-12 10:13:19.8605|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/{A69CFE30-74AE-4374-96EA-DE4ACD33D71F}/data/2 "Data Uploaded"
2024-10-12 10:13:19.8609|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/{A69CFE30-74AE-4374-96EA-DE4ACD33D71F}/data/3 "Data Downloaded"
2024-10-12 10:13:19.8609|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/{A69CFE30-74AE-4374-96EA-DE4ACD33D71F}/throughput/7 "Upload Speed"
2024-10-12 10:13:19.8609|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/{A69CFE30-74AE-4374-96EA-DE4ACD33D71F}/throughput/8 "Download Speed"
2024-10-12 10:13:19.8609|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/{A69CFE30-74AE-4374-96EA-DE4ACD33D71F}/load/1 "Network Utilization"
`

ernestodiaztech, 2024-10-12:
After I posted this I remembered what I did. I installed LibreHardwareMonitor.

ernestodiaztech, 2024-10-12:
Bruh, I'm dumb. I forgot the "install" when using the command with Powershell

nickbabcock, 2024-10-19:
Haha glad to see you figured it out.

closed 2024-10-19 by nickbabcock