nickbabcock/OhmGraphite #488: Supporting external LHM dll

FalenDemon, 2025-04-06:
Hello,

Thanks for your work! Recently I've updated my PC hardware and was suprised, that installed LibreHardwareMonitor can't read all the sensors from MB and CPU (B850 and Ryzen 9800X3D).
After a some time of researching, found this dll's: https://github.com/Alcolawl/LibreHardwareMonitor/releases using it solves wrong/missing readings from my hardware.
Th question is, could it be used with OhmGraphite somehow?

nickbabcock, 2025-04-07:
Interesting use case. Can you try the "unbundled" version of ohmgraphite, which should let you drop in that custom dll.

https://github.com/nickbabcock/OhmGraphite/actions/runs/14321211853

FalenDemon, 2025-04-08:
> Interesting use case. Can you try the "unbundled" version of ohmgraphite, which should let you drop in that custom dll.
> 
> https://github.com/nickbabcock/OhmGraphite/actions/runs/14321211853

Thanks for the chance to test. But no luck, after replacing LibreHardwareMonitorLib.dll from my link above.
CMD output:
`PS C:\Portable Soft\OhmGraphite-0.35-unbundled> .\OhmGraphite.exe start
Unhandled exception: System.ServiceProcess.TimeoutException: The operation requested for service 'OhmGraphite' has not been completed within the specified time interval.
   at System.ServiceProcess.ServiceController.WaitForStatus(ServiceControllerStatus desiredStatus, TimeSpan timeout)
   at OhmGraphite.OhmCli.<>c__DisplayClass1_0.<Execute>b__4(InvocationContext context) in D:\a\OhmGraphite\OhmGraphite\OhmGraphite\OhmCli.cs:line 77
   at System.CommandLine.Invocation.AnonymousCommandHandler.Invoke(InvocationContext context)
   at System.CommandLine.Invocation.AnonymousCommandHandler.InvokeAsync(InvocationContext context)
   at System.CommandLine.Invocation.InvocationPipeline.<>c__DisplayClass4_0.<<BuildInvocationChain>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass17_0.<<UseParseErrorReporting>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass12_0.<<UseHelp>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass22_0.<<UseVersionOption>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass19_0.<<UseTypoCorrections>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<UseSuggestDirective>b__18_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass16_0.<<UseParseDirective>b__0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c.<<RegisterWithDotnetSuggest>b__5_0>d.MoveNext()
--- End of stack trace from previous location ---
   at System.CommandLine.Builder.CommandLineBuilderExtensions.<>c__DisplayClass8_0.<<UseExceptionHandler>b__0>d.MoveNext()`

And it works with included LibreHardwareMonitorLib.dll.

nickbabcock, 2025-04-09:
That's a bummer. Kinda looks like the error message may be obscured if running as a service.

I wonder what the error message would be if you ran `.\OhmGraphite.exe run` instead of `.\OhmGraphite.exe start`

FalenDemon, 2025-04-09:
> I wonder what the error message would be if you ran `.\OhmGraphite.exe run`

I tried. No messages at all. Just new black window for a half of second appears and gone. In task manager ohm process not running.
`PS C:\Portable Soft\OhmGraphite-0.35-unbundled> .\OhmGraphite.exe run
PS C:\Portable Soft\OhmGraphite-0.35-unbundled>`

nickbabcock, 2025-04-22:
I'll close this for now, as there may be some technical hurdles. But anyone is welcome to chime in in the future as to how this might be supported.

closed 2025-04-22 by nickbabcock

nickbabcock, 2025-04-30:
I believe with this functionality merged into LibreHardwareMonitor and with the last update of OhmGraphite, this is underlying reason is now solved too