nickbabcock/OhmGraphite #490: Metric ohm_hdd_factor

rinasmichael, 2025-04-16:
This is a log:
_2025-04-14 10:59:43.2377|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/2 "CPU Core #1"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/3 "CPU Core #2"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/4 "CPU Core #3"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/5 "CPU Core #4"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/0 "CPU Total"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/load/1 "CPU Core Max"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/0 "Package"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/1 "Cores (Average)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/2 "Cores (Average Effective)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/3 "Core #1"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/4 "Core #1 (Effective)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/factor/0 "Core #1"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/1 "Core #1 (SMU)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/voltage/2 "Core #1 VID"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/5 "Core #2"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/clock/6 "Core #2 (Effective)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/factor/1 "Core #2"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/power/2 "Core #2 (SMU)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/voltage/3 "Core #2 VID"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/temperature/2 "Core (Tctl/Tdie)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/0/voltage/0 "Core (SVI2 TFN)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/load/0 "CPU Total"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/load/1 "CPU Core Max"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/power/0 "Package"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/clock/1 "Cores (Average)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/clock/2 "Cores (Average Effective)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/clock/3 "Core #1"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/clock/4 "Core #1 (Effective)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/factor/0 "Core #1"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/power/1 "Core #1 (SMU)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/voltage/2 "Core #1 VID"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/clock/5 "Core #2"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/clock/6 "Core #2 (Effective)"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/factor/1 "Core #2"
2025-04-14 10:59:43.2413|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/power/2 "Core #2 (SMU)"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/voltage/3 "Core #2 VID"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/temperature/2 "Core (Tctl/Tdie)"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /amdcpu/1/voltage/0 "Core (SVI2 TFN)"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/0 "Memory Used"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/1 "Memory Available"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/0 "Memory"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/2 "Virtual Memory Used"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/data/3 "Virtual Memory Available"
2025-04-14 10:59:43.2542|INFO|OhmGraphite.SensorCollector|Sensor added: /ram/load/1 "Virtual Memory"
2025-04-14 10:59:44.0396|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/0 "Used Space"
2025-04-14 10:59:44.0396|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/31 "Read Activity"
2025-04-14 10:59:44.0396|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/32 "Write Activity"
2025-04-14 10:59:44.0396|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/33 "Total Activity"
2025-04-14 10:59:44.0396|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/throughput/34 "Read Rate"
2025-04-14 10:59:44.0396|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/throughput/35 "Write Rate"
2025-04-14 10:59:44.0662|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/%7B702F2AB6-A3ED-4E65-AF2B-0021E6445411%7D/data/2 "Data Uploaded"
2025-04-14 10:59:44.0668|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/%7B702F2AB6-A3ED-4E65-AF2B-0021E6445411%7D/data/3 "Data Downloaded"
2025-04-14 10:59:44.0668|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/%7B702F2AB6-A3ED-4E65-AF2B-0021E6445411%7D/throughput/7 "Upload Speed"
2025-04-14 10:59:44.0668|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/%7B702F2AB6-A3ED-4E65-AF2B-0021E6445411%7D/throughput/8 "Download Speed"
2025-04-14 10:59:44.0668|INFO|OhmGraphite.SensorCollector|Sensor added: /nic/%7B702F2AB6-A3ED-4E65-AF2B-0021E6445411%7D/load/1 "Network Utilization"
2025-04-16 08:29:08.5934|INFO|OhmGraphite.Worker|Prometheus port: 4445
2025-04-16 08:29:11.1791|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/control/0 "Fan #1"
2025-04-16 08:29:11.1818|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/control/1 "Fan #2"
2025-04-16 08:29:11.1818|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/control/2 "Fan #3"
2025-04-16 08:29:11.1818|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/control/3 "Fan #4"
2025-04-16 08:29:11.1818|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/control/4 "Fan #5"
2025-04-16 08:29:11.1818|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/control/5 "Fan #6"
2025-04-16 08:29:11.1818|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/control/6 "Fan #7"
2025-04-16 08:29:11.3116|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/2 "CPU Core #1 Thread #1"
2025-04-16 08:29:11.3116|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/3 "CPU Core #1 Thread #2"
2025-04-16 08:29:11.3116|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/4 "CPU Core #2 Thread #1"
2025-04-16 08:29:11.3116|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/5 "CPU Core #2 Thread #2"
2025-04-16 08:29:11.3116|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/6 "CPU Core #3 Thread #1"
2025-04-16 08:29:11.3116|INFO|OhmGraphite.SensorCollector|Sensor added: /intelcpu/0/load/7 "CPU Core #3 Thread #2"
[... 81 lines cut ...]
2025-04-16 08:30:53.2220|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/fan/0 "Fan #1"
2025-04-16 08:30:53.2220|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/fan/1 "Fan #2"
2025-04-16 08:30:53.2220|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/fan/2 "Fan #3"
2025-04-16 08:30:53.2220|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/fan/3 "Fan #4"
2025-04-16 08:30:53.2220|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/fan/4 "Fan #5"
2025-04-16 08:30:53.2220|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/fan/5 "Fan #6"
2025-04-16 08:30:53.2220|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/nct6798d/0/fan/6 "Fan #7"
2025-04-16 08:30:53.3050|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ipmi/0/voltage/0 "+VCCIO_0"
2025-04-16 08:30:53.3050|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ipmi/0/voltage/1 "+12V"
[... cut ...]
2025-04-16 08:30:53.3066|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ipmi/0/fan/2 "FRNT_FAN3"
2025-04-16 08:30:53.3066|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ipmi/0/fan/3 "FRNT_FAN4"
2025-04-16 08:30:53.3066|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ipmi/0/fan/4 "REAR_FAN1"
2025-04-16 08:30:53.3066|INFO|OhmGraphite.SensorCollector|Sensor added: /lpc/ipmi/0/fan/5 "CPU_FAN1"_

What needs to be done to make the metric Ohm_hdd_factor work? 
I've already checked it on three machines, it doesn't work on any of them.

rinasmichael, 2025-04-16:
ohm_hdd_factor is empty metric on client's. 
What needs to be done for this metric to appear?
ohm_hdd_factor{instance="$instance",hw_instance="$hdd"}
What is it ohm_hdd_factor?

nickbabcock, 2025-04-17:
Before I can help, please let me know your backend where you are sending this data. OhmGraphite doesn't store any data on the client.

You can try increasing the log level and see if that helps you. https://github.com/nickbabcock/OhmGraphite?tab=readme-ov-file#debugging-tips

rinasmichael, 2025-04-17:
my debug log on client:
[OhmGraphite.log](https://github.com/user-attachments/files/19794486/OhmGraphite.log)
my backend is prometheus+grafana
this is a dashboard. 
[Ohm Windows Desktop-1744894900754.json](https://github.com/user-attachments/files/19794721/Ohm.Windows.Desktop-1744894900754.json)

nickbabcock, 2025-04-21:
I see. What kind of disks are these? I see an older issue mentioning something about that metric not appearing for Sata SSDs: https://github.com/nickbabcock/OhmGraphite/issues/375

Can you look in the LibreHardwareMonitor to see if they are reported there https://github.com/nickbabcock/OhmGraphite/issues/375#issuecomment-1585629120

Sata HDD should just work (https://github.com/nickbabcock/OhmGraphite/issues/119#issuecomment-597034963) but I don't have one on hand to test right now.

rinasmichael, 2025-04-23:
> I see. What kind of disks are these? I see an older issue mentioning something about that metric not appearing for Sata SSDs: [#375](https://github.com/nickbabcock/OhmGraphite/issues/375)
> 
> Can you look in the LibreHardwareMonitor to see if they are reported there [#375 (comment)](https://github.com/nickbabcock/OhmGraphite/issues/375#issuecomment-1585629120)
> 
> Sata HDD should just work ([#119 (comment)](https://github.com/nickbabcock/OhmGraphite/issues/119#issuecomment-597034963)) but I don't have one on hand to test right now.

[image]

nickbabcock, 2025-04-30:
What might help narrow down the problem is some extra logging. Can you try out this build and show me the output of the "Hardware Added" log line?

https://github.com/nickbabcock/OhmGraphite/pull/495
https://github.com/nickbabcock/OhmGraphite/actions/runs/14753572051?pr=495

closed 2025-09-24 by nickbabcock

eabugda, 2025-09-30:
So if the metric is not displayed in librehardwaremonitor, we should not expect it here in ohm? Just trying to understand the root issue

nickbabcock, 2025-09-30:
> So if the metric is not displayed in librehardwaremonitor, we should not expect it here in ohm? 

Correct 🙂

nickbabcock, 2025-09-30:
To clarify the hdd factors are only observable when you save report in librehardwaremonitor and ohmgraphite exposes these as continuous sensors.

nickbabcock, 2025-10-01:
Ah, this is a sata hard drive. Unless something changed, these should be supported out of the box, according to myself 5 years ago: https://github.com/nickbabcock/OhmGraphite/issues/119#issuecomment-597034963

As for the logging. That made it into the latest build and whenever you run in debug mode you'll see lines like:

```
Hardware added: /vram (Type: Memory, Class: VirtualMemory)
```

eabugda, 2025-10-02:
Great, here's the snippet of my startup log that shows anything `hdd`; I don't see any `factor`s anywhere in the log.

```
2025-10-02 15:20:49.4560|DEBUG|OhmGraphite.SensorCollector|Hardware added: /hdd/1 (Type: Storage, Class: StorageGeneric)
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/load/0 "Used Space"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/load/31 "Read Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/load/32 "Write Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/load/33 "Total Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/throughput/34 "Read Rate"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/1/throughput/35 "Write Rate"
2025-10-02 15:20:49.4560|DEBUG|OhmGraphite.SensorCollector|Hardware added: /hdd/2 (Type: Storage, Class: StorageGeneric)
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/2/load/0 "Used Space"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/2/load/31 "Read Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/2/load/32 "Write Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/2/load/33 "Total Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/2/throughput/34 "Read Rate"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/2/throughput/35 "Write Rate"
2025-10-02 15:20:49.4560|DEBUG|OhmGraphite.SensorCollector|Hardware added: /hdd/0 (Type: Storage, Class: StorageGeneric)
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/0 "Used Space"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/31 "Read Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/32 "Write Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/load/33 "Total Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/throughput/34 "Read Rate"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/0/throughput/35 "Write Rate"
2025-10-02 15:20:49.4560|DEBUG|OhmGraphite.SensorCollector|Hardware added: /hdd/4 (Type: Storage, Class: StorageGeneric)
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/4/load/0 "Used Space"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/4/load/31 "Read Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/4/load/32 "Write Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/4/load/33 "Total Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/4/throughput/34 "Read Rate"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/4/throughput/35 "Write Rate"
2025-10-02 15:20:49.4560|DEBUG|OhmGraphite.SensorCollector|Hardware added: /hdd/5 (Type: Storage, Class: GenericHardDisk)
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/5/temperature/0 "Temperature"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/5/load/0 "Used Space"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/5/load/31 "Read Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/5/load/32 "Write Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/5/load/33 "Total Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/5/throughput/34 "Read Rate"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/5/throughput/35 "Write Rate"
2025-10-02 15:20:49.4560|DEBUG|OhmGraphite.SensorCollector|Hardware added: /hdd/3 (Type: Storage, Class: StorageGeneric)
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/3/load/0 "Used Space"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/3/load/31 "Read Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/3/load/32 "Write Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/3/load/33 "Total Activity"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/3/throughput/34 "Read Rate"
2025-10-02 15:20:49.4560|INFO|OhmGraphite.SensorCollector|Sensor added: /hdd/3/throughput/35 "Write Rate"
```

nickbabcock, 2025-10-04:
Can you try out the artifact in https://github.com/nickbabcock/OhmGraphite/pull/525 ? I added some logging that should help us triage the issue 🙂 

I'm curious what version of pawnio is being detected.

eabugda, 2025-10-07:
Here's what was detected; I can provide the full log if required.
`2025-10-07 14:00:03.5701|INFO|OhmGraphite.SensorCollector|Detected PawnIo version (2.0.0.0) installed`