nickbabcock/OhmGraphite #492: How to get RAM bytes

dellingo, 2025-04-28:
Could someone share, how can i get the amount of RAM and how much of it is used in bytes not only in percentage?
# HELP ohm_ram_load_percent Metric reported by open hardware sensor
# TYPE ohm_ram_load_percent gauge
ohm_ram_load_percent{hardware="Generic Memory",sensor="Memory",hw_instance="ram"} 75.42823791503906
ohm_ram_load_percent{hardware="Generic Memory",sensor="Virtual Memory",hw_instance="ram"} 45.605525970458984

nickbabcock, 2025-04-30:
You don't see it something like:

```
# HELP ohm_ram_bytes Metric reported by open hardware sensor
# TYPE ohm_ram_bytes gauge
ohm_ram_bytes{hardware="Generic Memory",sensor="Virtual Memory Used",hw_instance="ram"} 66877284352
ohm_ram_bytes{hardware="Generic Memory",sensor="Memory Available",hw_instance="ram"} 13507907584
ohm_ram_bytes{hardware="Generic Memory",sensor="Memory Used",hw_instance="ram"} 55131828224
ohm_ram_bytes{hardware="Generic Memory",sensor="Virtual Memory Available",hw_instance="ram"} 29679738880
```

dellingo, 2025-05-02:
Omg im blind. 🫣
Thanx allot!

closed 2025-05-02 by dellingo